X360Tools/PicoFlasher #1: falcon

mynameistakenhowcanthatb, 2022-02-18:
having problems with falcon JRunner kicks saying picoflasher needs a update

jasper 100% fine with picoflasher

mynameistakenhowcanthatb, 2022-02-23:
System.TimeoutException: The operation has timed out.
   at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
   at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count)
   at System.IO.Ports.SerialPort.Read(Byte[] buffer, Int32 offset, Int32 count)
   at JRunner.PicoFlasher.RecvUInt32(SerialPort serial)
   at JRunner.PicoFlasher.<>c__DisplayClass20_0.<ReadNand>b__0()
   at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

RadioAct, 2022-02-26:
> having problems with falcon JRunner kicks saying picoflasher needs a update
> 
> jasper 100% fine with picoflasher

There is a bug in JRunner v3.1.1 version check and it will report the picoflasher needs an update if your running v3.0. Downgrade to 2.0 and you'll be good. JRunner will be updated at some point I'm sure, I'v seen mentions that there is a discord somewhere for JRunner, might check there for info on an update.

> System.TimeoutException: The operation has timed out.
> at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
> at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count)
> at System.IO.Ports.SerialPort.Read(Byte[] buffer, Int32 offset, Int32 count)
> at JRunner.PicoFlasher.RecvUInt32(SerialPort serial)
> at JRunner.PicoFlasher.<>c__DisplayClass20_0.b__0()
> at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
> at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
> at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
> at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
> at System.Threading.ThreadHelper.ThreadStart()

I am having this exact same issue on two separate jasper boards. I can get it to read a random small number of blocks (~0-30) then the picofreezes and stops servicing the USB endpoint correctly and windows throws the timeout message. I have to powercycle the pico to get it to respond again. I am waiting for delivery of a couple more picos to test if my board is damaged, If I have the same results I will attempt to run picoflasher in a debugger and see whats going on.

mynameistakenhowcanthatb, 2022-02-26:
i seen some one with the problem as well from long ago jasper boards v1 am not sure if you need to do something for it to work? been trying on v1 and can not get a read from it .. https://www.se7ensins.com/forums/threads/jasper-jrunner-can-t-read-nand.1597222/

RadioAct, 2022-02-28:
Jrunner has been updated now and supports the picoflasher 3.0 FW.

I tired some other pico boards and got some different results. Two other boards were able to read and write this Jasper consistently. I am not sure what the difference is but I am going to have to do some more investigation.

I also tried it on a Jasper with 512mb and the first pico won't read it either without freezing. The two other picos can read it but during writing they also freeze up. I am starting to suspect signal integrity issues with my setup as the failures are always at random points. The serial data lines may benefit from some termination resistors, I am going to get an oscilloscope setup and check out how those look.

Working now to get my pico debugger working and see if I can find out what happens when the pico freeze up.

ShawMerlin, 2022-02-28:
I have an issue with a Corona 4gb. I have it hooked up correctly by the
diagrams by weekend modder but jrunner gives and error and crashes when I
click read

Wiring is good as I'm able to use the xflasher 360 to read it, but the pico
doesn't work. I have the latest Jr runner pro from your github and also V3
on the pico itself.

Any ideas?

Thanks,
-Merlin Shaw

mynameistakenhowcanthatb, 2022-03-01:
Jasper with 512mb  works fine for me .
[image]

RadioAct, 2022-03-02:
Yea I can only get about 200 block written before my setup crashes on this Jasper. I know there is something with my setup that is the problem, just not sure exactly what it is. 
I am still trying to get a debugger working on the pico, I started with picoprobe thinking it would be simple but turns not not so easy. I have some time to work on it today and I'm thinking I should switch to Jlink and use that debugger instead.

RadioAct, 2022-03-03:
I have made some progress getting some data on the problem I am having. I was able to get the picoflasher running on a debugger and I have narrowed down the problem some. 

The issue seems to be with the USB side of the picoflasher. I have not seen any problems with reading from the xbox side, it is always failing on the USB CDC side. I turned on debugging with tinyUSB and was going over a log of the messages and what I see is that the USB transfers stop occurring on the endpoint and the cdc_tx buffer fills up. The code path then enters a bypass of stream() waiting for room in the buffer. I do not know when the USB transfers from endpoint stop happening, it could be something from the host side or the device side. The endpoint must not be responding to the USB host as after some time windows will i/o fault the com port. I'm not all that familiar with the USB process so I have some research to do in order to continue.

Here is a capture from the debug, sorry for the length:

> USBD Xfer Complete on EP 82 with 532 bytes
  CDC xfer callback
CDC_TX buffer level: 0x2000
stream_emc @ offset: 0x511
xbox_nand_read_block sucess
  Queue EP 82 with 532 bytes ...
result sent to usb cdc_write
USBD Xfer Complete on EP 82 with 532 bytes
  CDC xfer callback
CDC_TX buffer level: 0x2000
stream_emc @ offset: 0x512
xbox_nand_read_block sucess
  Queue EP 82 with 532 bytes ...
result sent to usb cdc_write
USBD Xfer Complete on EP 82 with 532 bytes
  CDC xfer callback
CDC_TX buffer level: 0x2000
stream_emc @ offset: 0x513
xbox_nand_read_block sucess
  Queue EP 82 with 532 bytes ...
result sent to usb cdc_write
CDC_TX buffer level: 0x2000
stream_emc @ offset: 0x514
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x1dec
stream_emc @ offset: 0x515
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x1bd8
stream_emc @ offset: 0x516
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x19c4
stream_emc @ offset: 0x517
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x17b0
stream_emc @ offset: 0x518
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x159c
stream_emc @ offset: 0x519
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x1388
stream_emc @ offset: 0x51a
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x1174
stream_emc @ offset: 0x51b
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0xf60
stream_emc @ offset: 0x51c
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0xd4c
stream_emc @ offset: 0x51d
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0xb38
stream_emc @ offset: 0x51e
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x924
stream_emc @ offset: 0x51f
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x710
stream_emc @ offset: 0x520
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x4fc
stream_emc @ offset: 0x521
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0x2e8
stream_emc @ offset: 0x522
xbox_nand_read_block sucess
result sent to usb cdc_write
CDC_TX buffer level: 0xd4
CDC_TX buffer is full
CDC_TX buffer level: 0xd4
CDC_TX buffer is full

For those following along at home, the interesting bits are these: 

> USBD Xfer Complete on EP 82 with 532 bytes
  CDC xfer callback

Those come from the tinyUSB stack. They show the USB host reading data queued from the endpoint. When they stop occurring the buffer fills up and the code enters endless loop of waiting for buffer space.

Now to find out why.

RadioAct, 2022-03-04:
Problem may be a fixed issue, will test with latest tinyUSB stack.

[https://github.com/adafruit/circuitpython/issues/4190](url)