X360Tools/PicoFlasher #4: Version 3.0 firmware isn't persistent on Pico

Pringleman83, 2022-03-15:
I've tried on two brand new Picos. The firmware uploads as it should and the PicoFlasher is visible in J-Runner. When I disconnect the Pico from the PC and reconnect, J-Runner no longer detects it.

When I install the version 2.0 firmware (following the same process), it works as expected.

Pringleman83, 2022-03-16:
On giving this some thought, the issue may not be persistence, but something else. When initially programmed and while connected, the pico works fine. When disconnected and reconnected, J-Runner doesn't detect it (but the pico isn't detected as a flash drive either).

xxS1NN3Rxx, 2022-03-21:
also having the same issue working on 3, 2 flashed fine, 3rd is having same issue as above, V2 is seen but outdated & V3 doesnt show up after a disconnect.

guilloplay, 2022-06-14:
i have the same problem, any solution for reconect and detect?

pacman362, 2022-07-05:
I think the issue is with the Pico itself im on my second Pico now and this is having problems too (not seing consul) in j-runner with extras ! this happened with my other Pico which I returned as faulty Now this one is doing it too ? I have another which should arrive today so I will connect it to the same Xbox and see if it works as expected, If it does then there must be a problem with Pico's going down after a few uses with J runner ?.

pacman362, 2022-07-11:
Ok update have had delivery of the 3rd pico and as before used it once to do a corona worked fine disconnected it went to do another and BOOOM no consul found as with both my other pico's ?? this has to be something with the Uf2 ?? becouse all other operations seem fine ie: can reflash UF2 file can flash other files all as norma, J-runner sees the pico no problems bt will NOT read anything from any xbox just consul not found in the output screen of j-runner ! this needs to be looked at by someone with the skills th debug what is happening to these picos after 1-2 uses.

NotPhoton, 2022-08-01:
> 

Where did you purchase your pico from? I'm thinking there are just fake boards coming from Amazon

pacman362, 2022-08-01:
They both of the semi-dead ones were from Amazon ! The one from PI hut is
fine.

NotPhoton, 2022-08-01:
> They both of the semi-dead ones were from Amazon ! The one from PI hut is
> fine.
> I think you could be right with the theory of fake ones on Amazon.

Seems like it's just these Amazon ones then! The other person I asked also bought theirs from Amazon, did you get yours from Seeed Studio?

pacman362, 2022-08-01:
Mine was from Frenove.EU

NotPhoton, 2022-08-01:
> Mine was from Frenove.EU

Hmm, freenove should be legitimate. Well, I've got 3 new Pico's from adafruit on the way, so I will compare those to my board when they arrive.

pacman362, 2022-08-01:
Yes I noticed the one from the pi hut was a darker board !

NotPhoton, 2022-08-01:
> Yes I noticed the one from the pi hut was a darker board !

Did the RP2040 chip in the middle of the board have the raspberry logo on it? I saw a couple forum posts about fake chips without the logo on them, just curious because mine does have it.

pacman362, 2022-08-01:
Left is the Pi hut (good one)right is the duff one actually not a lot
different after furthe examination and both have the raspberry logo.?

fabio296, 2022-08-02:
bom dia comprei o meu pico na alixpress mas nao reconhece na corona 4gb console nao reconhecido ..meu tem todas caracteristicas normal do pico ..que pode ser ????

NotPhoton, 2022-08-10:
> I have the same problem with my Rasberry Pico, When I get starting reading the NAND , sometimes at 25% or 50% reading the memory NAND , it stops, and I lose the firmware in the Pico. 
>  I have a Corona 4 GB but with the small chip memory.  I have never been anybody in Videos working with that kind of memory with the picoflash, in all of videos the people use the 4GB big chip memory. Did you check the picoflash with the 4GB small chip memory? I suspect that picoflash is not compatible with the small 4G memory NAND in Corona

Mine was also a corona 4gb with that type of memory, if you flip the board over there's a larger chip underneath. It really just sounds like you've got a faulty Pico, seems like there's just all sorts of issues coming from these faulty/fake picos

Soundtoxin, 2023-07-24:
I've run into this with two different Picos bought in-person at Micro Center that appear totally genuine to me.

pacman362, 2023-07-24:
Yep I have had another on go dead on me after 1 use I really do not
understand this ? I am using my reliable but slow JR programmer now.

NotPhoton, 2023-07-24:
> I've run into this with two different Picos bought in-person at Micro Center that appear totally genuine to me.

Now that is super odd, I know for sure micro center sells genuine Picos! Now wondering if its just down to different manufacturers/revisions of the RP2040. My Pico is still able to flash just fine, and I bought mine from adafruit.

pacman362, 2023-07-24:
Really weird as one of mine was from a reputable dealer for Rasberry ?? All of my pico,s still act as if they are fine , J runner sees them fine but just will not find the xbox.

xS1NN3Rx, 2023-07-26:
> I've run into this with two different Picos bought in-person at Micro Center that appear totally genuine to me.

i bought mine over a year ago from micro and had same issue.
2 differnt type chips in the batch, one with a dark logo one with a light logo.
put pressure on the chip as you plug it into the computer, if it connects you had a chip flow issue

pacman362, 2023-07-26:
Mine are fine as far as booting to computer as said in my earlier post ! They just seem to NOT communicate after 1-2 uses with J-runner they are picked up by the programme fine but will not communicate with the Xboxes ??

pacman362, 2023-08-06:
Ok Update, fitted both my suspected FAKE Pico's with GENUINE RP2040's And im pleased to report they both work perfectly ! So in conclusion I suspect that the chips where indeed FAKE ! So all of you out there with dead Pico's Order yourselfs some new RP2040's and get reworking to bring them back from the dead for your RGH work.

jiansong, 2023-08-18:
> Ok Update, fitted both my suspected FAKE Pico's with GENUINE RP2040's And im pleased to report they both work perfectly ! So in conclusion I suspect that the chips where indeed FAKE ! So all of you [...]

Let me guess, the not working one (or the "FAKE" one) has the "1921 - 5.2" on the backside, the constantly working one has "1921 - 5.13" on the backside. 

If this is true to most of you, then it's very likely related to the different batch of the boards.

pacman362, 2023-08-18:
Nope both of my repaired ones have : oc_136 94V

jiansong, 2023-08-18:
> Nope both of my repaired ones have : oc_136 94V

I mean the numbers following beneath that, should be two lines with smaller text like
```
1234
12.34 
```
see[ #17 ](https://github.com/X360Tools/PicoFlasher/issues/17#issuecomment-1236401926)

I got two boards, they have same number the first line "1921", I guess it's the code of the manufacture. 
They have different on the second line, one is "5.2", another is "5.13", could be the date of the production, don't know.

The "5.2" has faded text on the chip, the "5.13" has very clear text on the chip. 
The "5.2" has this exact problem, the "5.13" don't.

One detail is, I can't remember if the "5.2" was working correctly before soldering, the "5.13" is still new, and don't have this issue.

pacman362, 2023-08-18:
Only other text on these is 3221. And 1.9

pacman362, 2023-08-18:
Just noticed my older repaired one has 3421. 6.13

ryt51V, 2024-01-20:
I have had the same problem here.  Using a Raspberry Pi Pico bought from the Pi Hut (UK) in December 2023, the version 3.0 firmware does not persist after replugging the Pico, whereas the version 2.0 firmware persists just fine.  (I was modding a Trinity at the time, so I was able to mod it with the 2.0 firmware without issue).

I have since bought one of the "clone" Picos from AliExpress - specifically "TZT" type "RP2040 Type-C 16MB" with a black PCB, which appears to have a similar pinout to the original Picos (particularly the 7 pins needed for PicoFlasher, which appear to be exactly the same).  I avoided the purple PCB variants as they appear to have a very different pinout.

On this clone device, the version 3.0 firmware does persist, and is picked up by J-Runner just fine.  However I have not tested any actual reads or flashes with the clone as I'm not currently in the process of modding any consoles.

HVR88, 2024-04-06:
3 picos here, all legit. Two work and one has the problem described above. All same batch.

All work with other firmware for an unrelated project.