X360Tools/PicoFlasher #9: Falcon xbox no consul found.

pacman362, 2022-06-21:
I have a problem with a Falcon 360 running the latest dashboard but NOT being found by J-runner and Pico flasher ? I have just done another exact same Falcon and all is well ? but this one just shows up as CONSUL NOT FOUND x000000 (or something like this) in the output of J-runner????

Anyone had this issue before any words of advise.

PS: solder joints are fine I am skilled in p.c.b soldering.

Shinjikuro95, 2022-06-21:
To be honest i had only work on trinity and corona, but there was one time when a corona made something similar and those are bad news cause it could mean that the nand its corrupted or damadge. On my case, the nand had been damadge with water by a previous owner so it was only matter of time. You should check the resistors around the nand.

Is not much but maybe it will lead you on the right path?

Shinjikuro95, 2022-06-22:
The case i mentioned, the xbox boot up just fine also, but the nand couldn't be read of use so the internal storage was of no use in my case and i couldn't modify the nand

pacman362, 2022-06-23:
Thanks again for the reply, I have just opened another Falcon and have the exact same issue so this rules our the caps or resistors, I have ordered another Pico for today so I will try with that tonight if this does not work I'm going to start a fresh with another laptop using a fresh operating system.

Shinjikuro95, 2022-06-23:
Alright, i hope to find out if you figure this out cause it could help in the future. Good luck!

pacman362, 2022-06-23:
I will update if I find a solution

pacman362, 2022-06-24:
Ok quick update I ordered another Pico just in case and it turns out this has sorted the problems? So there must be some dodgy Picos on Amazon it did 1 Xbox and then I had this problem as above ! The new one has a darker coloured board as well which leads me to believe their may be fakes floating around ?

closed 2022-06-27 by pacman362

pacman362, 2022-10-11:
Thanks for the reply I will test components later tonight, The Xbox boots
fine though?.