X360Tools/PicoFlasher #12: Pico dying after 1-2 uses with j-runner

pacman362, 2022-07-11:
Ok update have had delivery of the 3rd pico and as before used it once to do a corona worked fine disconnected it went to do another and BOOOM no consul found as with both my other pico's ?? this has to be something with the Uf2 ?? becouse all other operations seem fine ie: can reflash UF2 file can flash other files all as normal , J-runner sees the pico no problems but will NOT read anything from any xbox just consul not found in the output screen of j-runner ! this needs to be looked at by someone with the skills to debug what is happening to these picos after 1-2 uses.

Androxilogin, 2022-11-08:
I had this happen after flashing a few as well. I just switched back to my JR programmer for typical installs and use the Pico for 4GB reads for the time being. It's not ideal having to switch back and forth but those 4GB reads are a pain otherwise. Seems like it's burning something out on the board. Maybe a power spike on the order the power is applied. 

I've seen some say to apply the power to the Pi first but that seemed to be where my issues began. I had much better luck applying power to the console first then the Pico. I would also assume removing the console power first would allow the voltage to drop properly since it wouldn't be back fed the spike. Just my current theory, anyway.

scottroadnight, 2023-01-03:
This has been the case for me for quite sometime now. Jr programmer works fine but no matter what board I have the picoflasher says not found with either all 0's or all D's. Worked well for 6 consoles to start though.

Might have to try another Pico just for the sake of it.

m-i-k-e-e, 2023-01-06:
Same issue for me, at some point it managed to read the nand. But after unplugging it, it went back to its original failure