X360Tools/PicoFlasher #16: Windows not detecting device after reconnecting.

Adran-Marit, 2022-07-21:
Upon initially flashing picoflasher 3.0 to my Raspberry Pi Pico, it is detected by windows and J-runner correctly and I am able to use it as such.

The problem occurs once I've disconnected the pico and reconnect it to my computer. 
When I plug it back in windows doesn't detect the device unless I enter it back into flashing mode to flash a different .uf2 file to it.
While I can work around this, by reflashing the firmware it is a pain to have to do it every time.

Important Information
Device: Raspberry Pi Pico
Firmware: PicoFlasher 3.0
Operating System: Windows 10 Pro (Ver 21H1 OS Build 19043.1826)
Software: J-Runner with Extras 3.1.5b & J-Runner Pro v3.2.0

xxS1NN3Rxx, 2022-08-11:
several others had this issue, thread was hyjacked by console not found complaints then marked as invalid.
i fiddled around and worked out if you put your finger on the main chip the board will work (ive been in contact with pico and they say its not a defect)
have had multiple genuine boards do this and pressing the chip in different places will cause each board to work normally until you stop pressing the chip.

i would be personally willing to ship one of my boards to somebody in Aust for testing to find the fault

Adran-Marit, 2022-08-11:
I'm not sure this is a hardware issue on my end, I've spoken to balika and will be dumping the spi when I get the chance. 

I find it odd version 2 of firmware works with no issues and version 3 doesn't. 

I'm in aus but can't really do hardware testing atm

xxS1NN3Rxx, 2022-08-13:
i bought 20 between our 2 aussie official suppliers had had same issue in both batches- 6/20.

i also found it extremely odd V2.0 works but 3 doesnt.
if your doing further testing of dumping ect this would be something to note as we have the identical issue of v2 working v3 does not and windows doesn't see it after a disconnect.
im more than happy to provide a video to show how consistent pressing the chip works on any of my 6 boards with the same problem, it was dumb luck how i found this worked, held the board different and plugged it in and it popped up, retried and it didnt, after frustration of seeing it pop up once i kept fiddling untill i discovered this, tested other boards and to my surprise they all worked if pressed but to make it really weird they all had different places on the chip that required pressure, one was top left, one was bottom left ect.
i honestly cant explain it and try not to think about it as it hurts my brain, for all i know my thumb was acting as a heat shield or making the chip contact. (one of the company i bought from seen the video and said i possibly caused static damage, but my work area and conditions should have avoided that)

Adran-Marit, 2022-08-13:
I just tried holding the chip in a few different places with no luck, I'm wondering if they could be fake chips like [this ](https://github.com/X360Tools/PicoFlasher/issues/17#issuecomment-1206629469)post in one of the other threads.

mine is hard to read I would be interested to see if the 6 you have are the same, or they are different. I definitely need to order another pico or two though for other projects

xxS1NN3Rxx, 2022-08-14:
further inspection, can confirm ALL the picos that failed are very faint marking on chips, ALL working picos have a very white logo.
both suppliers are official resellers, so maybe different batches rather than fake/real.

i still am confused about pressure on chip as all of my boards worked that way.

mr-bridger, 2022-08-16:
i have this same issue, my pico was from a authorised seller and does not look fake, it will not reconnect to windows at all with version 3 once disconnected, i have to reflash to get it to be recognise.  pressing the chip makes no difference.

Using version 2 i can reconnect, i hear the usb connect sound and the com port is immediately recognised,

Adran-Marit, 2022-08-18:
> 

Does yours have a very faint logo on it or a nice bright one?

mr-bridger, 2022-08-18:
Nice bright logo. looks perfect.

Adran-Marit, 2022-08-21:
> Nice bright logo. looks perfect.
Huh, could you maybe try holding at points on the chip like what was suggested earlier?

mr-bridger, 2022-08-21:
As I said in my first message, pressing the chip makes no difference.

Adran-Marit, 2022-08-21:
Ah derp I was tired ahaha

Odd so yours has bright printing on the chip? Curious indeed

retrofixconsolas, 2022-10-04:
II had the same problem and it occurred to me to press hard on the main IC to see if it was a cold solder and it work. I did a little reflow and it hasn't given me any problems. In case someone else tries it and know if that is the cause of the problem.

Diegocpx, 2022-11-13:
Hey guys! im from brasil. Mi pico have a very faint logo, so its fake right? =( ive pressed on chip (discredited) and it worked. you have to press on silver chip near the RB logo