X360Tools/PicoFlasher #24: RP2040-Zero support in the future?

Androxilogin, 2022-11-08:
Not sure if there are enough pins or how to convert the code if even possible, but I thought it would be cool to use one of these little things in place of the nano.

liquidzorch, 2023-06-05:
Hello. I have changed the pins in the source code, and I can confirm my rp2040-zero works with both 16mb and 4gb nands. Here is the file compiled for the zero just delete the ".pdf" from the file

The GPIO for 16mb are swapped from and to:
16 - 0
17 - 1
18 - 2
19 - 3
20 - 4
21 - 5
gnd - gnd (note the zero only has one gnd pin)

The GPIO for 4gb are swapped from and to:
6 - 26
7 - 27
8 - 28
9 - 29
21 - 5 (same as 16mb)
gnd - gnd

I just used my zero to program a 16mb and a 4gb today, both came out great, although the 4gb is too slow for my preference.
The LED was also reprogrammed, but does not work (I never got it working on the pico either), but besides the led, all seems to work.

[PicoFlasher3_zero.uf2.pdf](https://github.com/X360Tools/PicoFlasher/files/11655010/PicoFlasher3_zero.uf2.pdf)

Androxilogin, 2023-06-05:
Oh wow! Thanks for sharing! I'll have to buy another console and test it out!

everyonesweird, 2023-07-21:
Thanks. I was looking for this also. Using the RP2040 zero for Switch. Thought I could use it for XBOX also. Thanks!

Androxilogin, 2023-07-26:
> Hello. I have changed the pins in the source code, and I can confirm my rp2040-zero works with both 16mb and 4gb nands. Here is the file compiled for the zero just delete the ".pdf" from the file [...]

I tried this last night on a 16mb Corona V1 and it worked great! I have yet to try it on a 4GB. I threw together this diagram 

[IMG]https://imgur.com/Qa4cYRM.jpg] 

to help myself along without taking note that it was a Trinity and wired the blue and green wires backwards but it still worked that way for whatever reason. But this is how it should be wired according to what you have provided above so I left it that way.

liquidzorch, 2023-07-26:
Yeah, I can't remember what pins 4 and 5 do, but when wirtting one has to go low (and from testing is always low on the picoflasher when connected), which prevents the console from booting, the other, I can't remember what it does, but it probably goes low as well when reading/writing, and thats probably why it worked swapped. I know you can't swap them on the jr programmer, because one pin does not stay low. Any way, glad it has worked for you guys. I am using one myself, but I use a normal sd card reader for the 4gb, since its about 100x faster than the picoflasher.

Androxilogin, 2023-07-26:
That's weird since I found the regular Pico to be a lot faster and more reliable than the SD tool. Although it is still in beta and I don't really know all that much about the hardware behind the Pico/w to begin with. I was still getting the grasp of Arduinos when this all came out of nowhere. I'll have to give it a try when I get a hold of one. As long as it's not like ten minutes to read I'm sure I'd be fine with it.

liquidzorch, 2023-07-26:
With a normal Sd reader, it takes me about 5 seconds to read/write. With the pico flasher (regular, not RP2040-zero), it took well over 5 minutes, plus I would sometimes get an error randomly and would have to start again. Maybe it was just my connections being too long or something. But I was doing it that way for awhile, since my sd reader was acting up, and I could never get a read in under 5 minutes.

Androxilogin, 2023-08-06:
Thank you again for taking on this contribution. For this version, I have designed a case to share. I used female pin headers with the pins trimmed halfway (about 2mm) in which you feed through the bottom to the appropriate connections on the Pico. Pictured is my prototype so it differs slightly from the one provided in the package below. Ground connection is centered for both pinheaders. Also provided is a diagram for anyone interested. 
[image]

[image]

[PicoFlasher-Zero.zip](https://github.com/X360Tools/PicoFlasher/files/12270795/PicoFlasher-Zero.zip)

kelpyss, 2024-10-03:
It's not working for me, after writing the code on rp2040-zero the PC doesn't recognize the device 😢

sablezis, 2025-05-20:
Is there a scheme for sonus?
https://youtu.be/BN3pHJF77eg?si=hwvjTbaKCsxWUXmY

Flamelord626, 2025-06-11:
anybody mind shareing the sourcecode for the modification for the RP2040-Zero i want to make some changes

warbeast44, 2025-07-12:
Has anything like this been done for flashing glitch chips like matrix with the pico zero 
 Dirtypico on the zero