X360Tools/PicoFlasher #30: Possibility to program glitch chips such as ACE v3 aswell?

The-Lord-of-Owls, 2023-04-13:
Is pico flasher able to also program chips like the ACE v3 aswell? if not I'd like to know if this is something that could possibly be added as it would really be nice to eliminate the need entirely for programming equipment if we can also program those mod chips aswell as the xbox nand.

nikdiklic, 2023-08-06:
I believe it is possible, but since the appearance of RGH3 not many people are willing to make such modification, because RGH3 makes glitch chips obsolete and unnecessary. I suggest you get more familiar with RGH3

chackAJMCPE, 2023-09-25:
So how do we do it? (I accidentially broke cpu_pll_bypass pad.)
I would need to do an s-rgh

The-Lord-of-Owls, 2023-09-28:
> I believe it is possible, but since the appearance of RGH3 not many people are willing to make such modification, because RGH3 makes glitch chips obsolete and unnecessary. I suggest you get more familiar with RGH3

RGH3 from what I've heard doesn't like to play well with dual nand on certain boards and with certain nand images. Thats kinda a deal breaker for me since I always dual nand

chackAJMCPE, 2024-01-14:
I have finded a way. https://www.se7ensins.com/forums/threads/j-runner-with-extras-dirtypico360-program-timing-files-to-glitch-chips-using-a-pico.1844370/

The-Lord-of-Owls, 2024-12-27:
Sorry for the really late reply but this helps quite a bit, thank you!

closed 2024-12-27 by The-Lord-of-Owls