X360Tools/PicoFlasher #31: Windows does not detect device after flashing

ProTx220, 2023-04-13:
tried version 2 and 3
I have pico with usb type c and 16 mb of memory

SevenPayne, 2023-04-18:
+ 1

rcbraga08, 2023-05-25:
Would you help me? When I put Picoflash on Pi Pico, it disappears and does not install the COM port, do you know what could be happening? If I put another random file that I downloaded from the UF2 internet, it works and changes to the COM port, only with Picoflash that doesn't do this procedure.

ProTx220, 2023-05-30:
> Would you help me? When I put Picoflash on Pi Pico, it disappears and does not install the COM port, do you know what could be happening? If I put another random file that I downloaded from the UF2 internet, it works and changes to the COM port, only with Picoflash that doesn't do this procedure.

Do you have the original PICO?