X360Tools/PicoFlasher #39: NAND 4GB TOSHIBA INCOMPATIBLE WITH PICO FLASHER

TheBannedBoy809, 2024-01-20:
I use RGH 3.0 on several Xbox 360s, and I noticed that some Pico Flasher had compatibility problems, especially on 4GB consoles, so I started to notice that the TOSHIBA brand uses another form of communication that is not available on the Pico Flasher, for example all other brands you can read with 5 pads through an SD Card Reader or the Pico Flasher that uses the same 5 pad connection, but the TOSHIBA requires all 8 pads to be recognized in the JRUNNER, so if I take any SD CARD and put it only the 5 pads will not be recognized, and therefore the same error will occur on the Pico Flasher, only with 8 pins a reading is possible.

Dear developer, would it be possible to implement the other 3 pads for Pico Flasher programming or unfortunately in these cases is it necessary to use an SD CARD reader?

[image]

Andrei1744, 2024-01-24:
Same issue here! If the dev can see this, please fix

KonzolozZ, 2024-01-24:
Remove R1R1 resistor on the bottom side when reading and writeing. Xell can boot witouth R1R1.

Andrei1744, 2024-01-25:
> Remove R1R1 resistor on the bottom side when reading and writeing. Xell can boot witouth R1R1.

Dude you are a mf savior!!!

TheBannedBoy809, 2024-01-26:
> Remova o resistor R1R1 na parte inferior ao ler e escrever. Xell pode inicializar sem R1R1.

> Remove R1R1 resistor on the bottom side when reading and writing. Xell can boot without R1R1.

Thank you my friend, I removed R1R1 and managed to read it correctly with Pico Flasher, thank you very much for the tip.

Best of all, it is not necessary to replace the R1R1 component. Thank you very much, the Pico flasher is much better than the SD card for reading and writing as it rarely gives errors while the SD card usually ends up giving errors when recording.

closed 2024-01-26 by TheBannedBoy809

KonzolozZ, 2024-01-26:
> Thank you my friend, I removed R1R1 and managed to read it correctly with Pico Flasher, thank you very much for the tip.
>
> Best of all, it is not necessary to replace the R1R1 component. Thank you [...]

After read write need to install back R1R1 resistor. Console boot instable to RGH without R1R1 ( if Toshiba NAND + WB2KB).
Also need to remove R1R1 if use SD Card adapter.