X360Tools/PicoFlasher #41: Using Ubuntu to put the uf2 file on the pico causes the pico to not show up anymore

GalaxiumNation, 2024-04-04:
So as described, I drag and drop the uf2 file from downloads to the pico and when it copies over it disconects and doesnt reconect. at all. even when I unplug it and replug it in it doesn't show up. I have to hold the button on it an then plug it in for it to show up again, but that means I have to put the file bac on the pico, again. Tried v2 as well, same thing. I do have the option to use windows 7 but I'd rather use Ubuntu 20.04 for this because it's much faster than my other pcs.

Help apreciated!

dueiriel, 2024-04-28:
That's exactly how it works: 

you plug it while holding the button to be able to write firmware. 
Drag and drop the file, it gets written then the pico disconnects.

After the firmware is written you are only able to "see" it again by holding the button.

Please don't try to mod if did not do this little research first.

GalaxiumNation, 2024-04-29:
Yes, I realized that. I got it working and have proto going as well.

closed 2026-11-02 by GalaxiumNation