X360Tools/PicoFlasher #43: WILL NOT Read Corona v6 4gb NAND, Wiring is correct, and neat.

BluntSlapp, 2024-04-10:
As the title states. J-runner finds my console fine, even starts to read it, but it fails about 3 seconds in with a different error code each time. 

I've tried literally everything I can think of, this NAND is also completely fine, I was using it this morning. 

I've checked my wiring, I've checked, and double checked the wiring diagram, rgh3 is installed correctly, etc.

I'm really pulling at stings to find a reason as to why this would happen, so I thought I'd let yall know.

Wapitiii, 2024-06-22:
Same issue for me with a v6, picoflasher worked with my v2 corona. Any idea why that could happen? I dont think my wiring is bad.

roomert, 2024-09-23:
Sorry for the late comment.

Had this issue myself, the Pico is unable to read the 4GB consoles as it functions as an SPI flasher and its storage is WAAAAAY too small to handle the data coming off of the NAND chip. You need to get an actual programmer, XFlasher is the goto one.

huck0031, 2024-12-08:
Having the same issue here, what's strange is I managed to r/w a Corona 4gb Nand _perfectly_ fine a couple weeks ago, but it seems the current board I'm working on is just not having it (production date . I have re soldered dozens of times and have tried multiple Picos, I will update when I receive my SD Tool as I have decided to go that route for this particular system.

This particular board I am having problems with is a X857330-003-4GB

Below is the last few lines of my console.log file:
```

=========================================================================
J-Runner with Extras
Session: 12/07/2024 23:13:33
Version: 3.3.0 r4
Status: Up to date

WARNING - Working Folder!
Your working folder is not empty, click Show Working Folder to view its contents

Checking Console...
Corona: 4GB (eMMC connected)

Checking Console...
Corona: 4GB (eMMC connected)

Reading Nand to C:\Users\olivi\OneDrive\Documents\xboxTools\J-Runner-with-Extras\output\nanddump1.bin
Error: 230E3672

Read Successful!

Initializing nanddump1.bin, please wait...
Glitch2 selected
Nand Initialization Finished

Cancelled

=========================================================================
J-Runner with Extras
Session: 12/07/2024 23:17:05
Version: 3.3.0 r4
Status: Up to date

WARNING - Working Folder!
Your working folder is not empty, click Show Working Folder to view its contents

Checking Console...
Corona: 4GB (eMMC connected)

Checking Console...
Corona: 4GB (eMMC connected)

Reading Nand to C:\Users\olivi\OneDrive\Documents\xboxTools\J-Runner-with-Extras\output\nanddump1.bin
Error: 27DB12BF

Read Successful!

Initializing nanddump1.bin, please wait...
Glitch2 selected
Nand Initialization Finished
```