X360Tools/PicoFlasher #45: PICO FLASHER CAN'T BE DETECTED AGAIN

Nicklosss, 2024-07-26:
hello, could someone help me? i was trying to reuse my rpi pico but, after flashing it jrunner doesn't detect it, how did  i do wrong? hold on bootsel button and plug rpi into the windows pc, after that copy the .utf2 file and it close automatically the window, i launch jrunner but it doesn't recognize my pico flasher as the first time. thanks for the answers

gir3atingbrainz, 2024-07-26:
I had the same problem what worked for me was when I was holding the button after the drive pops up I enter the drive then let go of the button then copy over the utf2 file over

raider952, 2024-09-10:
Picoflasher with 160 mhz clocl frequency. I compiled from sources for fake pico. https://drive.google.com/file/d/19PPErDrT_yCMoBELtLvHjLc9glzMFO-c/view?usp=drive_link

chuckyto, 2024-11-27:
> Picoflasher with 160 mhz clock frequency. I compiled from sources for fake pico. https://drive.google.com/file/d/19PPErDrT_yCMoBELtLvHjLc9glzMFO-c/view?usp=drive_link

That solved the issue with my "Raspberry pico". I thought it was real, not fake...

Thank you!

evgnor86, 2025-01-23:
> Picoflasher with 160 mhz clock frequency. I compiled from sources for fake pico. https://drive.google.com/file/d/19PPErDrT_yCMoBELtLvHjLc9glzMFO-c/view?usp=drive_link

THANK YOU MAN!!!