X360Tools/PicoFlasher #46: PicoFlasher reads, but cannot write on Trinity

XanderStrike, 2024-07-27:
J-Runner is able to identify the console, read the NAND twice and successfully compare them, and create XeLL. No problems.

When I go to Write XeLL, it fails immediately with an IO error, and advises me to disconnect and reconnect.

Once it's reconnected, I can identify and read NAND not problem, but it still fails on write. 

It also fails the same way if I immediately try writing after connecting it.

I'm following this guide: https://xbox360hub.com/guides/rgh-3-guide/

Solder points are solid, I've checked and rechecked. I've done the RGH 3 wiring and NAND header wiring as described in the guide.

I've tried two separate Picos but they're identical. I've tried the V2 and V3 PicoFlasher firmware and there's no difference in behavior.

I don't know of any way to extract logs from the pico flasher but let me know if there's something that would be helpful and I'll do it.

<details>
<summary>Click for some pictures</summary>

[image]
[image]

The Pico (both Pico's I've tried are identical):
[image]
[image]
</details>

UrbisR, 2024-08-22:
I had the identical issue. I then copied my J-Runner folder to my desktop instead of my laptop, and ran it from there. It then worked and was able to write the NAND. Perhaps the laptop was not able to supply enough power through its USB port?

cogray86, 2024-08-24:
Any update on this? I'm having the same issue on 4GB Corona.