X360Tools/PicoFlasher #47: Picoflasher reading Corona 16mb as 4gb.

XenXenOfficial, 2024-08-19:
For some odd reason, Picoflasher is having a REALLY hard time running and figuring out my Corona 16mb console. It repeatedly reads it as 4gb, occasionally it gets it right, but a majority of the time it just ends in a "System.TimeoutException" which isn't too detailed of an error... Is there any ways of fixing this?

Ian-McCleary, 2024-08-25:
I can confirm i'm having the same issue. Checked the console storage before RGH3 and confirmed it was 16mb RAM + compared to reference photos. I followed connection order of open JRunner, console power, pico flasher into computer.
[image]

XenXenOfficial, 2024-08-27:
> Was going to comment saying Im having the same issue, but I just realized I did the 4gb wiring for a 16mb console since I left my wiring alone from the last one I modded. Same issue by chance?

I am unsure if it's the same issue, but I've redone my RGH3 wiring and I'm currently waiting for an Xflasher to come in, I've read online Picoflasher is pretty unreliable, especially with Coronas. I'll give ya an update once if I make any progress!