X360Tools/PicoFlasher #50: Raspberry Pi Pico 2 (RP2350) future compatibility

Belucih, 2024-09-08:
Hi!

I have tried to “install” PicoFlasher on my new Raspberry Pi Pico 2 with RP2350 chip, but it doesn’t work. My local shop no longer has stock from the old Pico 1 with RP2040 chip. Are you planning to support this board too?

Thank you for your work!:)

Meow, 2024-11-29:
I've successfully compiled it for pico 2 and it worked as-is with J-Runner, I only had to change cmake config to make that happen, the vscode plugin did all the work for me, I just added tinyusb_device and tinyusb_board as dependencies. If you wish I could send you the compiled binary, but you can probably get it to compile yourself fairly easily using vscode + pico dev plugins for it.

MrMeugh, 2024-12-13:
> If you wish I could send you the compiled binary, 

that compiled binary would be quite nice. I'm a newbie when it comes to using vscode and cmake, or compiling anything for pico

fishscrounger, 2024-12-14:
@Meow Oh wow, good shout, thanks! If you could send the binary, that would be great thanks. Otherwise, I'll have a go at compiling it myself

MasterCats98, 2024-12-14:
@Meow  pls i beg you can you drop the file? i can't do it alone. It would be so much appreciated.Thank you!

Meow, 2024-12-14:
Alright, so here's the entire project folder from my drive, I simply just used vscode pico tools to change build target to rp2350 and it worked. I am dumb and bought a pico 2 and really, really did not want to buy another pico just for that :p I do not claim that this is a good port, I literally simply just changed the build target.

The binary is located at `./PicoFlasher/build/PicoFlasher.uf2`. Only compatible with pico 2. Simply drag and drop into the pico 2 folder, it should immediately unmount the folder upon doing so.

**Use at your own risk, for educational purposes only. I am not to be held liable for any damages, losses or consequences that might occur by using this software.**

_Note: I did observe my pico 2 to overheat at one point when using it, but I think it was due to my soldering to the board being bad. I did successfully read NAND and then flash XELL using it. At no point was my pico 2 damaged._

https://drive.google.com/file/d/19eL80_dMqSUZoTclTjXj0t1ZLq-53kTz/view?usp=sharing

MasterCats98, 2024-12-14:
@Meow thank you so much! You saved my day

fishscrounger, 2024-12-14:
Hmm, not working on my Pico 2 here. I guess I'll try and build it myself. Thanks all the same!

Meow, 2024-12-14:
> Hmm, not working on my Pico 2 here. I guess I'll try and build it myself. Thanks all the same!

When using it with J-Runner, you might have to plug it in a few times to get J-Runner to recognize it. Like unplug and plug back in until it works.

fishscrounger, 2024-12-14:
> 
> When using it with J-Runner, you might have to plug it in a few times to get J-Runner to recognize it. Like unplug and plug back in until it works.

My Pico 2 doesn't even accept the binary

Meow, 2024-12-14:
> > When using it with J-Runner, you might have to plug it in a few times to get J-Runner to recognize it. Like unplug and plug back in until it works.
> 
> My Pico 2 doesn't even accept the binary

Just checked the archive I sent and mine does. Are you dragging the .uf2 file?

fishscrounger, 2024-12-14:
> Just checked the archive I sent and mine does. Are you dragging the .uf2 file and holding BOOTSEL key when connecting your pico?

Well...it turns out that I was somehow in a different directory and copied a different PicoFlasher.uf2 file! I have no idea how I managed to do that 😆 Thanks for the help, I can stop configuring cmake now.

MasterCats98, 2024-12-15:
@Meow  hi, i have another problem. i started to mod the xbox 360 like a month ago with a pi pico w. unfortunately i coudn't continue bcz jrunner would just stop working.it finds the console, but as i click read nand, jrunner crush, stops, and return functioning just bcs there was a time limit. i bought so a rasberry pi pico 2, but i have the same problem. i drop here a screenshot, hope you have a solution, thx so much for the help.
[image]

Meow, 2024-12-15:
> @Meow hi, i have another problem. i started to mod the xbox 360 like a month ago with a pi pico w. unfortunately i coudn't continue bcz jrunner would just stop working.it finds the console, but as i click read nand, jrunner crush, stops, and return functioning just bcs there was a time limit. i bought so a rasberry pi pico 2, but i have the same problem. i drop here a screenshot, hope you have a solution, thx so much for the help. ![IMG_20241215_145744](https://private-user-images.githubusercontent.com/191707661/395867293-52630df1-723d-4b1d-9011-68b86a31a1a2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyOTcxOTEsIm5iZiI6MTczNDI5Njg5MSwicGF0aCI6Ii8xOTE3MDc2NjEvMzk1ODY3MjkzLTUyNjMwZGYxLTcyM2QtNGIxZC05MDExLTY4Yjg2YTMxYTFhMi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxNVQyMTA4MTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lZjIwODAzMTY1NzMzOWE1M2M4NDIyYWNiMzZkNTQwZTEzMDFkNTdmZTY2NGIwZjFkNzUxMGM3YWM1Yjg4Y2M1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.bCAQgHdsKOjem7naTae8MAEVXecJBA0RZN9PXt0pd7I)

Hi! I unfortunately do not offer support. The compiled binary / software is provided as-is without any guarantees or support of any kind. I do not know enough about J-Runner or Xbox Modding myself to really help you. Ask on the forums, that'd be your best bet.

NoizyDTM, 2025-05-10:
@Meow im so lost on this bought a pico 2 thinking it would be fine cause i cant find pico original and this confused me even more lmao

MasterCats98, 2025-05-10:
> [@Meow](https://github.com/Meow) im so lost on this bought a pico 2 thinking it would be fine cause i cant find pico original and this confused me even more lmao

Simply you can't use the normal picoflasher.uf2 with a Pico 2. But @Meow created (or got from someone else, I really don't remember) the uf2 file shaped perfectly for the Pico 2. Unfortunately I broke my Xbox (4gb version), and I can't tell if the file could have worked...

NoizyDTM, 2025-05-10:
@MasterCats98 im just trying to understand how i get the file to test it myself i really dont know to much about coding or anything like that so im just lost jumping into this

Alan1508, 2025-05-20:
i lack braincells so i do not understand what is happening, but based on context clues:

you can get the picoflasher port for the pico2 by opening Meow's Google Drive folder and go through PicoFlasher>build>PicoFlasher.uf2
then you just copy and paste the uf2 into the boot loader and that's how i got it working. thanks