X360Tools/PicoFlasher #54: Too high system clock on RPi Pico compatible board

prodeveloper0, 2024-12-29:
I found that maximum system clock written in document is133MHz not 266MHz.
I think the reason makes problems not recognizing or not working some Pico board (including compatible board)

There is PicoFlasher binary adjusted system clock. I tested with 16MB NAND, 4GB eMMC is not tested.
https://github.com/prodeveloper0/PicoFlasher/releases/tag/3.2

Toetje585, 2025-08-01:
This is quite a big one this could have caused so many issues people where having.
Good find.... <3

Sakitoshi, 2026-03-01:
it may also be a good idea to lower the voltage back to 1.1v too since the cpu isn't being overclocked anymore.
change `vreg_set_voltage(VREG_VOLTAGE_1_30);` to `vreg_set_voltage(VREG_VOLTAGE_1_10);`